mitxela/kicad-round-tracks #1: Program crashes and becomes unresponsive

rgbbytes, 2021-01-31:
I'd appreciate if you could elaborate on the extents and use cases for this program. In relatively complex designs (100 components), I ran this and it just made KiCAD unresponsive after 30 seconds and nothing would snap it out of that. Was this not meant to handle larger systems? Any ideas or experiences on crashing?

mitxela, 2021-01-31:
It just takes a while to run. There is no loading bar at the moment.

You can set the number of passes to be 1, and then run it several times, and it will give the same result. A single pass shouldn't take too long.

The design I was testing it on is not enormous, but it's not small either - it has 132 components and about 800 vias. Three passes takes about 22 seconds to run on my machine.

rgbbytes, 2021-01-31:
I gave it half an hour and nothing but a frozen GUI. Maybe it's a difference of... processor.

mitxela, 2021-01-31:
Might be something in your design is crashing it then. If you can try and find what's causing it, it would be helpful. You could do a binary search by deleting half the design and seeing if it still crashes, then deleting half of what remains, and so on.

rgbbytes, 2021-01-31:
I'll do that sometime this week and report back.

antogg, 2021-11-13:
I'm also facing the same issue.
It seems to hang KiCad right after the dialog is shown. It doesn't seem to matter whether I clicked on RUN or just closed the dialog.

Application: KiCad
Version: (5.1.12-1-10_14), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.39.2
Platform: macOS Catalina Version 10.15.7 (Build 19H1519), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
    Boost: 1.76.0
    OpenCASCADE Technology: 7.5.3
    Curl: 7.54.0
    Compiler: Clang 10.0.1 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON

mitxela, 2021-11-20:
Interesting. I tried the plugin on the file you shared and it worked fine for me.

[image]

I tested on Arch linux and Windows, I'm unable to test it on macOS right now.

If it crashed before you even hit run, that might mean it's something in the GUI that's causing problems. Did you run the teardrops plugin OK? Have you tried any other kicad plugins?

Your version info has USE_WX_GRAPHICS_CONTEXT=ON, whereas both of my installs have it off. The docs say that's an experimental flag, but they might be out of date. Did you compile this build yourself?

antogg, 2021-11-23:
I'm using precompiled KiCad, installed via Homebrew package manager https://brew.sh/

I did install the teardrop plugin and another plugin IBOM https://github.com/openscopeproject/InteractiveHtmlBom and they both ran fine.

I too am guessing there's something in the GUI that's causing it to hang hard, but haven't been able to get it setup to run from the command line. Running python that's packaged with KiCad is rather finicky..

pondahai, 2021-11-24:
MacOS 12.0.1
Kicad 5.1.10-1-10_14

GUI freeze

message from command line:
`Traceback (most recent call last):
  File "/Applications/KiCad/kicad.app/Contents/SharedSupport/scripting/plugins/kicad-round-tracks/round_tracks_action.py", line 79, in run
    Popen(['pcbnew', new_name])
  File "/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/Applications/KiCad/kicad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
`

mitxela, 2021-11-25:
Does it work if you untick "create a new file" ?

antogg, 2021-11-25:
In my case it still hangs the main UI and I had to force-quit KiCad afterwards.

Here's what I did:
1. I opened KiCad with the above project.
2. I clicked on the pcbnew icon to open the layout editor
3. I refreshed the plugins via "Tools"->"External Plugins..."->"Refresh Plugins" menu
4. I ran the Kicad-Round-Tracks plugin via "Tools"->"External Plugins..."->"Round the tracks (new file)" menu
5. The dialog opened and I can uncheck the "create new file"

Up until just before the dialog appeared pcbnew was still responsive (moving my mouse moved the cross hair), but almost immediately after the dialog appeared KiCad hung. I can close the dialog or click on "Run" and KiCad would stay unresponsive until I force-quit it and restart it.

One observation: if I selected "new file" and clicked on "Run" it did create the rounded track version of the board with the name "<board_name>.kicad_pcb-rounded" that I can rename to something.kicad_pcb and open with pcbnew after I restarted KiCad.

lhondareyte, 2021-12-26:
Hi, same problem here on macos 11.6.1: new pcb board is written but GUI hang. I suspect that the GUI become unresponsive because this plugin try to open a second instance of pcb_new. This is not allowed on macos: you can have only one instance at a time.
On macos, the plugin should create a new board and exit (new kicad_pcb file without new project file)

lhondareyte, 2021-12-26:
Hi, same problem here on macos 11.6.1: new pcb board is written but GUI hang. 
Even if I close the plugin window without action, GUI is hang. It seems that there is problem at exit.

Note that on macos, the option "open the newly created file" should no be available : you can have only one instance of a running programm (here, pcb-new) at a time.

ScaredyCat, 2022-01-16:
I have macos 11.6.2 with the same issue.  Just hangs the entire kicad instance.

mitxela, 2022-01-17:
I will look into this when I can. If anyone wants to lend me a macbook that would speed things up (I just wasted several hours trying to run macos in a vm)

ScaredyCat, 2022-01-19:
> I will look into this when I can. If anyone wants to lend me a macbook that would speed things up (I just wasted several hours trying to run macos in a vm)

Have a look at  https://github.com/foxlet/macOS-Simple-KVM Worked for me on a spare machine, no issues at all.

mitxela, 2022-01-21:
That should fix the GUI hang, please try it out.

The "open the newly created file" error posted by pondahai is a separate problem. There is probably a way to fix it but to be honest I've always thought that part was janky. You can launch multiple instances of pcbnew but it isn't really supposed to be used that way, I might get rid of that entirely, maybe make the default to run in-place and offer to save a backup file.

ScaredyCat, 2022-01-24:
Doesn't hang for me any more, but doesn't run either. Crashes complaining that 'BOARD_DESIGN_SETTINGS' object has no attribute 'GetNetClasses' (line 46 round_tracks_action.py , self.board.GetDesignSettings.GetNetClasses.NetClasses() )

mitxela, 2022-01-24:
Sorry about that - please try the branch "kicad-5" - I just started porting everything to kicad 6 which has broken compatibility.

ScaredyCat, 2022-01-24:
Thanks, that fixes that problem - it now rins and produces curved/rounded tracks. 

The only thing that remains is that if I run it a third time, it locks up pcbnew. First 2 times there's no issue on a really simple board. Third time always locks up.

closed 2022-01-26 by mitxela

ScaredyCat, 2022-01-27:
> I'm not sure about that one. I've cherry-picked the commit that possibly fixes the process not ending correctly.
> 
> The intended use case is to keep the unrounded source file, and run the plugin to generate a rounded output file. If you're running it repeatedly on the same file, since each pass basically doubles the number of tracks, running three times with three passes could quite easily cause it to lock up by having too many tracks.

Ok, that's fine. I can work with that as a process. build my board, rebuild/add remove stuff then generate rounded tracks last minute to seperate pcb file. 

> Since KiCad 6 is released, I think I'm going to focus on the KiCad 6 version for now.

Absolutely, fully understand this.