mitxela/kicad-round-tracks #3: segfault on close

mitxela, 2022-01-29:
KiCad sometimes segfaults when you close pcbnew after running the plugin.

It's not a huge deal because you're already trying to close the program, but would be good to know why this happens.

trholding, 2022-01-31:
I've been test driving this. Pretty nice. However I haven't seen a segfault so far.

mitxela, 2022-02-01:
It seems the segfault only happens when tracks were deleted. In the plugin this only happens when two corners are close enough that the original track would be reduced to zero length. I've tried a few different ways of deleting tracks and they all seem to cause the same problem. I suspect it might be a glitch in KiCad 6 itself.

trholding, 2022-02-02:
Can you send me a project to test this out? I couldn't reproduce. BTW I am using KiCad 6.01

trholding, 2022-02-02:
@mitxela Not related to this issue, but this came to my mind. There is another great plugin that I wish to use in conjunction with your plugin. That one is called stretch, the coolness factor with stretch is that it can output the board to SVG and one can do fancy stuff in Inkscape to edit tracks. The magic is that one can import that SVG back into kicad with the plugin. Right now it doesn't work as expected with KiCad 6 due to API changes. So if you are in for a challenge or maybe make it cross compatible with your round tracks plugin, have a look at this issue https://github.com/JarrettR/Stretch/issues/15 .

mitxela, 2022-02-03:
Simplest possible example: [test.kicad_pcb.zip](https://github.com/mitxela/kicad-round-tracks/files/7995261/test.kicad_pcb.zip) Round it with native arcs and radius of 2mm. The segfault only happens as you close pcbnew, so if you're not running a debug build it can be hard to spot. It's logged in dmesg / event viewer / whatever though.

Stretch looks interesting, I will have a play with it.

mitxela, 2022-02-12:
Another data point: I think this only happens if the board had unsaved changes. It might be to do with the "discard changes" dialog box.

I think this is outside my jurisdiction, so I've filed an issue with kicad: https://gitlab.com/kicad/code/kicad/-/issues/10827

mitxela, 2022-04-09:
This is fixed as of KiCad 6.0.4 - https://gitlab.com/kicad/code/kicad/-/merge_requests/1136

closed 2022-04-09 by mitxela