mitxela/kicad-round-tracks #8: "Better" handling for 3-point junctions

mr-september, 2022-04-24:
Not sure how to properly describe this, but basically a hole is opening up at 3-point junctions. I don't think this has practical consequences, but I also barely know anything about electronics to have any confidence in that.

[image]

[image]

mitxela, 2022-04-25:
I suppose we could fix this by making the plugin leave a short track in the middle to fill the hole, but that wouldn't be very satisfying. I think the only way to get decent T-junctions, especially if the tracks are different widths, would be to do it with zone fills.

I think I might add a checkbox to just ignore multi-point junctions. Then, if I feel up to it, I could add T-junction handling to the teardrop script.

tpimh, 2022-04-27:
I wonder if it's a viable solution to leave the junctions untouched by rounding plugin and handle them with teardrop plugin instead?