mitxela/kicad-round-tracks #9: Some T-connections don't get rounded !

FredM67, 2022-04-28:
Here's a piece of my PCB...
[image]

After calling the plugin, here's what I get:
[image]

Both blue T-Connections on the left and in the middle-right don't get round. The other ones (middle-left and the 2 on the right) are rounded properly... Strange !

mitxela, 2022-04-29:
Two reasons this can happen. First is that the long horizontal track is one piece, so where the vertical track touches it, it isn't considered a junction. You can fix it by splitting the long track, or just redrawing the vertical one. Usually, when you draw a T-junction KiCad will split the track properly, but not always, depending on how it was drawn.

The other way this can happen is if the ends of the track are close but not quite touching. To be considered a join, the track ends need to be within 10nm of each other. This can sometimes happen if you switch grid sizes. If you right click / properties on each track, you can see their start and end positions, maybe one of them is on the grid while the other has 0.001 offset or something.

FredM67, 2022-04-29:
> Two reasons this can happen. First is that the long horizontal track is one piece, so where the vertical track touches it, it isn't considered a junction. You can fix it by splitting the long track, [...]
Yes, it was btw intentional to have both variants just to analyse the problem. Now it's ok, I had to completely redraw these track.

> The other way this can happen is if the ends of the track are close but not quite touching. To be considered a join, the track ends need to be within 10nm of each other. This can sometimes happen if [...]
Yes, this point is quite tricky... a "magnet" feature should be added inside the function "Cleanup track" as it happens when drawing a track.

By the way, I encounter another problem with junctions like this one:
[image]

After rounding I get:
[image]
This leads the Teardrop plugin to do simply nothing there, because the arc is "passing through"

Same happen with another linear junction where a straight piece of track is created passing through the pad.

mitxela, 2022-04-29:
Yeah that's a known issue ( https://github.com/mitxela/kicad-round-tracks/issues/4 ) will fix it when I get a chance

closed 2022-04-30 by mitxela