mitxela/kicad-round-tracks #11: Add a progress bar

nm17, 2022-05-02:
Just a good thing to have. An ETA would be even better

mitxela, 2023-02-26:
I know this was originally requested because the program had crashed and it wasn't obvious, but I had a stab at a basic progress bar anyway. With native arcs it usually only takes a few seconds, but whatever.

The most annoying bit is that we can't show progress for rebuilding the zone fills. So the program appears to hang at that step. Also, if you have auto-refill zones enabled, I can't detect that and it will always refill them twice. The python interface is very limited in this regard, we could patch the KiCad source code but I'm not sure it's worth it.

dumblob, 2023-02-27:
>Also, if you have auto-refill zones enabled, I can't detect that and it will always refill them twice. The python interface is very limited in this regard

Could not we track the refill ourselves in our separate python data structures and then check it to avoid this double refill?

Yes, it would be fragile in case the API changes but I think it is worth it.

Thoughts?

mitxela, 2023-02-27:
> > Also, if you have auto-refill zones enabled, I can't detect that and it will always refill them twice. The python interface is very limited in this regard
> 
> Could not we track the refill ourselves in our separate python data structures and then check it to avoid this double refill?
This wouldn't make any difference. The (minor) problem is simply that after the plugin runs, the zones are refilled again.

Sounds like in 7.0.1 the auto-refill zones option will be disabled by default, so this is probably a non-issue.

closed 2023-03-11 by mitxela