mitxela/kicad-round-tracks #16: crash when running

aaronw2, 2022-06-12:
Traceback (most recent call last):
  File "/home/aaronw/.local/share/kicad/6.0/3rdparty/plugins/com_github_mitxela_kicad-round-tracks/round_tracks_action.py", line 141, in validate_all_data
    tested_val = float(self.netclasslist.GetTextValue(i, j))
RuntimeError: wrapped C/C++ object of type DataViewListCtrl has been deleted

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aaronw/.local/share/kicad/6.0/3rdparty/plugins/com_github_mitxela_kicad-round-tracks/round_tracks_action.py", line 104, in on_item_editing
    self.validate_all_data()
  File "/home/aaronw/.local/share/kicad/6.0/3rdparty/plugins/com_github_mitxela_kicad-round-tracks/round_tracks_action.py", line 145, in validate_all_data
    self.netclasslist.SetTextValue(str(RADIUS_DEFAULT), i, j)
RuntimeError: wrapped C/C++ object of type DataViewListCtrl has been deleted
Traceback (most recent call last):
  File "/home/aaronw/.local/share/kicad/6.0/3rdparty/plugins/com_github_mitxela_kicad-round-tracks/round_tracks_action.py", line 141, in validate_all_data
    tested_val = float(self.netclasslist.GetTextValue(i, j))
RuntimeError: wrapped C/C++ object of type DataViewListCtrl has been deleted

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aaronw/.local/share/kicad/6.0/3rdparty/plugins/com_github_mitxela_kicad-round-tracks/round_tracks_action.py", line 104, in on_item_editing
    self.validate_all_data()
  File "/home/aaronw/.local/share/kicad/6.0/3rdparty/plugins/com_github_mitxela_kicad-round-tracks/round_tracks_action.py", line 145, in validate_all_data
    self.netclasslist.SetTextValue(str(RADIUS_DEFAULT), i, j)
RuntimeError: wrapped C/C++ object of type DataViewListCtrl has been deleted
Segmentation fault (core dumped)

Application: KiCad PCB Editor

Version: (6.0.5-152-g7765e63f10), release build

Libraries:
	wxWidgets 3.1.7
	libcurl/7.66.0 OpenSSL/1.1.1d-fips zlib/1.2.11 libidn2/2.2.0 libpsl/0.20.1 (+libidn2/2.2.0) libssh/0.8.7/openssl/zlib nghttp2/1.40.0

Platform: Linux 5.3.18-150300.59.68-preempt x86_64, 64 bit, Little endian, wxGTK, , x11

Build Info:
	Date: Jun 11 2022 21:58:23
	wxWidgets: 3.1.7 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.75.0
	OCC: 7.6.1
	Curl: 7.66.0
	ngspice: 36
	Compiler: GCC 11.3.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON

Python 3.10.4 installed with latest wxPython

mitxela, 2022-06-12:
Could you let me know a little more about what caused this? Is it repeatable? Does it only happen on one board design? Does this also happen in 6.0.5 release?

aaronw2, 2022-06-12:
It's actually 2044fe591dcccbcf172302767d915e66cdd84446 but I cherry-picked a focus-stealing fix which should not impact this. I have not been able to recreate this. It has only happened once.

closed 2023-03-11 by mitxela