mitxela/kicad-round-tracks #19: running time almost doubles every time i hit run

alinelena, 2022-11-05:
i like to build incrementally the pcb and I round the tracks after each step.
what i noticed is that running time almost doubles every time.

this is on kicad 6

mitxela, 2022-11-05:
The intended use of the plugin is to keep the unrounded source file and generate a new rounded PCB from it. I haven't tested running the plugin repeatedly on the same design. If you are using subdivisions that will certainly cause problems.

If you select part of the board, the plugin will only operate on the selection, so that might be a helpful way forward if you don't want to keep a separate source file.

alinelena, 2023-02-22:
i went in the end for the new file every time...

closed 2023-02-22 by alinelena