mitxela/kicad-round-tracks #21: Script causes PCB to be renamed

aaronw2, 2022-12-20:
After running the script with KICad 6.0.10 the PCB is renamed from filename to filename-rounded. This causes problems when attempting to later edit the PCB since it loads the original filename which does not contain any changes made since running the script. I did not see this behavior with earlier versions of KiCad.

Steps to reproduce:

1. Open design in PCBNew
2. Run rounded tracks operation
3. Notice in title that filename is now filename-rounded
4. (optional) perform any other changes to the PCB
5. Save the output (File->Save or ^S)
6. Quit PCB editor
7. Reopen PCB editor
8. All recent changes are not present

It seems that after running this that saving to the filename now saves to filename-rounded instead of filename.

mitxela, 2022-12-20:
Do you have "Create new file" checked? That is the expected behaviour when it's ticked. It is partly a remnant from earlier versions - before the undo history worked properly, before it was possible to apply to just the selection, and especially before native arcs were added.

You should still be able to open the rounded file output in pcbnew, just by clicking it in the project manager.

It's not impossible a recent update changed the checkbox behaviour, I can't test right now though.

aaronw2, 2022-12-20:
My mistake. I had it set.

mitxela, 2022-12-20:
No worries

closed 2022-12-20 by mitxela