mitxela/kicad-round-tracks #22: Plugin not working on KiCad 7

ShaojieHe, 2023-02-14:
I download this plugin from KiCad PCM, upon use and showed this error when using this plugin.

[image]

mitxela, 2023-02-14:
I'm surprised the PCM even lets you download it as it's tagged only for KiCad 6, but yes there's some API changes that mean the plugin will need updating. I'm a bit too busy to do it right now but I'll get round to it eventually. Or if anyone else feels up for doing it feel free to submit a pull request.

ShaojieHe, 2023-02-14:
thanks for reply, Sorry I'm bad at python coding so I will leave it to others :)

mitxela, 2023-02-15:
Might as well leave this open so others can see it

alinelena, 2023-02-22:
I have tried today to see if I can update but somehow seems to be more complicated that expected... 

naively using self.board.GetAllNetClasses() results in 

[image]

mitxela, 2023-02-23:
The netclass stuff was rewritten and the api is now a bit broken. I have submitted a pull request here: https://gitlab.com/kicad/code/kicad/-/merge_requests/1533
Once that (or something similar) gets merged, the plugin will work again, following the changes I pushed yesterday.

benrogmans, 2023-02-23:
Thanks for submitting that PR! I hope it gets merged soon :-)

alinelena, 2023-02-25:
I can confirm.. kicad patched with the 1533 and git plugin works.

mitxela, 2023-02-25:
It's merged, it should also get cherry-picked for 7.0.1. I will publish a new release when that comes out.

mitxela, 2023-02-26:
The patch is part of the nightly builds, so the plugin should work with those. I just added some very basic progress bar stuff, it would be helpful if someone can test it out on Mac as I only have Linux and Windows here.

FrightRisk, 2023-02-27:
It doesn't work for me. Windows 11, KiCAD 7 official release. I tried the install from the plugin manager, and of course it didn't work. But I wanted to get the folder name of where it wanted it. Then I unzipped master, copied it over everything in that folder (C:\Users\Fred\Documents\Kicad\7.0\3rdparty\plugins\com_github_mitxela_kicad-round-tracks), deleted the cache folder and tried again. I loads the first screen, I check the box to make a new file, and tell it to run. Another box pops up saying "processing", but nothing happens. I can't close the window because "close" is grayed out. It is stuck at "processing". I have to load the task manager to terminate KiCAD to be able to get out of it.
Then I saw that I needed the nightly build of KiCAD, so got this: [kicad-nightly-r31999.79f13ea9c7-x86_64-lite.exe] from today. I just installed over top of the one in the 7.0 folder instead of having yet an other 7.99 folder. It acted like a new install asking what settings I wanted, but everything is there. The about version says 7.99. I get the same exact error:

[image]

alinelena, 2023-02-27:
tested in linux and works... are you sure you put the new files from git in the right place?

I used the release script to bundle them in.

FrightRisk, 2023-02-27:
Working! You were right. I went to uninstall and realized that even though I said to copy things to the 7.0 folder, I could see 7.99. So I looked in preferences to see path locations and it still changed 2 of the paths (3rdparty being one of them) to 7.99. I think that is a bug in Kicad. I changed the paths to point to 7.0 and it worked. Thank you. I LOVE this plugin ;)

endlessempathy, 2023-03-04:
Somehow plugin doesn't work on latest 7.99 nightly build. I mean it works but does nothing. I also tried creating new project. It just shows progress bar, then shows popup saying that work is done but tracks are still "unrounded".
[image]

mitxela, 2023-03-05:
It seems to still be working for me. I tried it on 7.99.0-342-gd714ff8549 with the latest version of round-tracks (c0a4fb8).

Can you try it on a very simple project? Also try running it with the scripting console open to see if there are any python errors. Otherwise maybe share a project where it's not working.

endlessempathy, 2023-03-05:
> 

Probably I'm just dumb and missing something, but I tried creating new project with just one trace and it still doing nothing.
I tried installing plugin (v1.2) from KiCad Plugin manager then changing files inside the folder to latest files from github and deleting __pycache__ folder, tried uninstalling that plugin and just unzipping current github repo (https://github.com/mitxela/kicad-round-tracks/commit/c0a4fb819697a50a7257ffe14afa4d7378cd3b88) into KiCad\7.99\3rdparty\plugins\kicad-round-tracks-master again. Getting no errors don't know what to do next. Should I delete 7.99 folder completely and try to reinstall KiCad?
[image]

endlessempathy, 2023-03-05:
> Should I delete 7.99 folder completely and try to reinstall KiCad?

Of course it worked 🙄 I fee so dumb now...
Anyway, thank you for your answer! Maybe if someone will encounter same problem they can see my msg.

FrightRisk, 2023-03-06:
I installed my KiCAD in the same 7.0 folder so I don't keep having to have a different folder (like 7.99) every time I install. But KiCAD leaves 2 of the folders pointing to the old folder name. No matter where you install it, you just have to make sure it is looking in the right place for the plugins

mitxela, 2023-03-11:
KiCad 7.0.1 is now released, you can download round tracks 1.4 from PCM.

closed 2023-03-11 by mitxela