mitxela/kicad-round-tracks #25: Not working in KiCad 7.0

scottywei, 2023-03-23:
Seems kicad-round-tracks won't work in KiCad 7.0, I got a project which was created in KiCad 6, and just upgraded to version 7 by my newly installed KiCad 7.0.1, I tried to get all my tracks rounded and it seems working, giving me the time it takes, but nothing happen to my PCB design, I tried creating a new project, placed some very simple routes in PCB design, but kicad-round-tracks still won't change anything to the PCB.

Version Info:
---
Application: KiCad x64 on x64

Version: 7.0.1, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.0.1
	FontConfig 2.14.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 11 2023 03:41:46
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	ngspice: 39
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON

mitxela, 2023-03-23:
What version of round tracks?

Can you run it again with the scripting console open (Tools > Scripting console) and see if there are any errors?

scottywei, 2023-03-23:
> What version of round tracks?
> 
> Can you run it again with the scripting console open (Tools > Scripting console) and see if there are any errors?

Thank you for replying. I'm using v1.4 of round track, and what command shall I suppose to typing in the console?

mitxela, 2023-03-23:
Just run the plugin while the scripting console is open, if there are errors it should show a popup.

scottywei, 2023-03-23:
Hi, I just solved the problem by totally remove my KiCad 7 installation, including the plugin folders and the setting folder in %APPDATA%/kicad/, then reinstall KiCad 7 and DO NOT import settings from KiCad 6, use a new default setting instead, then install the round track plugin, it works fine now.

scottywei, 2023-03-24:
Seems when I have kikit and kicad-round-tracks installed at the same time in KiCad 7, the kicad-round-tracks won't work, when I uninstall kikit and restart KiCad 7, the kicad-round-tracks will work agian.

mitxela, 2023-03-24:
Aha, well spotted. I'll look into it.

mitxela, 2023-03-24:
So, looks like the author of KiKit has made this compatibility layer called pcbnewTransition, which is implemented by fiddling with sys.modules -  https://github.com/yaqwsx/pcbnewTransition/blob/main/pcbnewTransition/transition.py

This isn't supposed to affect other plugins but I'm doing some fairly brittle checks like `type(track) == pcbnew.PCB_TRACK` which now fail with KiKit installed. It's kind of my fault I suppose (but kind of not) but at least it's an easy fix.

mitxela, 2023-03-25:
Fixed in 1.5

closed 2023-03-25 by mitxela