mitxela/kicad-round-tracks #26: Net inspector shows old values after smoothing

dzid26, 2023-11-29:
Net-inspector shows old values after smoothing - until I refill the Zones (B) or remove the zones (Ctrl+B). 

https://github.com/mitxela/kicad-round-tracks/assets/841061/b567d4de-e7e2-43e9-8954-af3a3db73999

I am on 7.99.

mitxela, 2023-11-30:
Sorry, KiCad nightly isn't supported - it's too much of a commitment to keep up with the changes while it's in flux.

From a cursory glance this doesn't seem to happen on 7.0, so you might be better off reporting this to the kicad devs.

dzid26, 2023-11-30:
I see on 7.0.9 net inspector doesn't update if it is left in the background when performing rounding.

I thought you may know why, since you are performing the zone refill already, 
https://github.com/mitxela/kicad-round-tracks/blob/master/round_tracks_action.py#L105 and yet it is still needed to refill them again manually.

mitxela, 2023-11-30:
I don't have time to look into it right now. I would guess it has nothing to do with zone filling, just that there is a hook to update the net inspector window when the board is edited, and running a python script doesn't trigger that hook.