mitxela/kicad-round-tracks #27: Keep out copper zones

besi, 2024-02-13:
Hello there. I am using the switch from jlcpcb (the one that's part of the basic parts), which has a copper keep out zone, which is a bit weird I admit.

Now the round tracks when applying the plugin are all over the front-lawn of the push button and DRC is not happy.

[image]

mitxela, 2024-02-15:
You have a few options, the easiest is just to manually correct the problem, or in your case you could probably just ignore the error. You can split tracks before rounding, and this will limit their radius. The rounded tracks will always pass through the midpoint of the original tracks.
As mentioned in #6 there are limitations in the KiCad python API that mean the plugin is blind to DRC errors, so this isn't something that can be easily fixed.

besi, 2024-02-15:
Thanks for the heads up