mitxela/kicad-round-tracks #29: The plugin takes an incredibly long time to round tracks in Kicad 8.0.8

umbertoragone, 2025-01-16:
I just updated from Kicad 8.0.7 to 8.0.8 (on macOS Sequoia 15.2, M1 Macbook Air 2020) and I tried opening a project saved in 8.0.7 that had the tracks rounded with your awesome plugin. When opening the .kicad_pcb file, it gives a beachball cursor for a solid couple of minutes, then loads the board, but when trying to refill the solid fills (copper pours), it just hangs at "Removing isolated copper islands".

I then tried removing the rounded tracks and the solid fills generate instantly, as expected. If I then try to generate the round tracks with your plugin, pcbnew (PCB Editor) just becomes unresponsive (with the beachball cursor) for a couple of minutes, then the rounded tracks appear. But when I go ahead and try to regenerate the solid fills, it still hangs at "Removing isolated copper islands" for a while (2-3 minutes), then it returns responsive as nothing happened.

I even tried reinstalling the plugin, but the issue persists.

Are there any logs I could share to diagnose and help fix this problem? Any help is highly appreciated.

mitxela, 2025-01-18:
To confirm, it sounds like it's not the plugin that's slowed down, but the copper fills that have become slow? And the same file used to open fine in the previous version? It is probably a regression in KiCad then, so it might be worth filing the issue there - https://gitlab.com/kicad/code/kicad/-/issues

umbertoragone, 2025-01-18:
Yes, the copper fills have become incredibly slow if there are rounded tracks on the board (up to 8-10 minutes wait). If there are no arcs/rounded tracks in the board design, it works instantly as expected. I'm not sure if something major changed from 8.0.7 to 8.0.8 that would prevent generating copper fills correctly.

If you (or somebody else) could try to replicate the issue, just to confirm it is not a problem on my machine only, I would appreciate it.