mitxela/kicad-round-tracks #32: Mixed class traces

dzid26, 2025-03-06:
Ground is Power class and it is located between differential pairs and it doesn't get rounded. 

[image]

I needed to remove the Power class from Gnd as a workaround.

mitxela, 2025-03-07:
Could you kindly share which version of kicad you're using (haven't had a chance to test on kicad 9 yet) and also share an example file of the problem, just delete everything on the board except that bit in the screenshot so I can test it.

dzid26, 2025-03-07:
It was in Kicad 9. 

I tried to reproduce in Kicad 8, but I couldn't . I think it is a matter of some board setting. 

I am attaching both. 
[Downloads.zip](https://github.com/user-attachments/files/19130718/Downloads.zip)

mitxela, 2025-03-08:
OK, looks like the project file has been corrupted or truncated. The netclasses tab shows blank for most of the fields for the power class:

[image]

I edited the kicad_pro file and pasted in the missing data from the v8 file you shared:
```
      {
        "bus_width": 12,
        "clearance": 0.1,
        "diff_pair_gap": 0.25,
        "diff_pair_via_gap": 0.25,
        "diff_pair_width": 0.1,
        "line_style": 0,
        "microvia_diameter": 0.3,
        "microvia_drill": 0.1,
        "name": "Power",
        "pcb_color": "rgba(0, 0, 0, 0.000)",
        "priority": 0,
        "schematic_color": "rgba(0, 0, 0, 0.000)",
        "track_width": 0.1,
        "via_diameter": 0.4,
        "via_drill": 0.15,
        "wire_width": 6
      }
```
(that's all of the same fields, plus the new "priority" field)

Then I opened the file in kicad 9 and it appears to work fine.

I could harden the plugin against this happening but the real question is why the fields got wiped to begin with. Is this a project that was started in kicad 8 and then updated to 9? Maybe the update system is buggy.

dzid26, 2025-03-08:
I can't remember if I started the project in Kicad 8. 

But in Kicad 9, when you create a new net class, fields are always empty!  I am not sure if that's on purpose. 

Additionally, even after I filled all the blank fields, the issues persists.
[image]

dzid26, 2025-04-22:
I tried to do some more changes to my project and encountered another issue, but it's probably related. 

I temporarily remove the classes as a workaround to the apply smoothing. After I close "Smoothing is finished (took 10 second)" window, Kicad takes 100% CPU (on multiple cores), ram usage increases and I need to kill it.  I think this workaround used to work last time, so something must have changed in my project. 

Attaching my 
[Dock-flex.zip](https://github.com/user-attachments/files/19846643/Dock-flex.zip)
in case you have time to look at it.

mitxela, 2025-06-07:
Hi, sorry it took me a while to get round to this.

With Dock-flex.zip it seems there's a problem on the right hand connector, under the middle ground pin there are over a thousand tracks of almost zero length:

[image]

I deleted those tracks and it seems to work fine, and takes a lot less than ten seconds to round. Do you know how those ended up in the design? I could potentially filter out any track shorter than a certain amount, but it would slow down rounding most designs that don't need that.

Will look into the netclass problem shortly.

mitxela, 2025-06-07:
In KiCad 9, it's possible for a net to have multiple netclasses.

If a netclass does not define every parameter, it will inherit from the Default class, and is considered to be in both netclasses. So the GND tracks are in both Power and Default. 

There's some discussion of this here: https://forum.kicad.info/t/v9-netclass-seems-to-always-add-default/59004

I'm not sure what the use case of multiple netclasses is, but if a netclass doesn't uniquely identify a group of tracks then it's less useful for grouping different track rounding amounts. I'll have a think about this.

In the mean time, you can work around this issue by defining every parameter for each netclass. It turns out you can show the hidden columns in the PCB editor window by right-clicking on the column header:

[image]

By filling in a value for Wire Thickness and Bus Thickness, the Power class became unique and the plugin runs normally again.

dzid26, 2025-06-07:
I think those zero length tracks were caused by Kicad 9's new multi track shove feature
https://youtu.be/i7EDQurbUtg?t=16. (I remember some issues where it was dividing straight segments when getting close to an obstacle).   
Alternatively, they were caused by unrounding feature - but I have no evidence. 

You are right - any missing class param is causing the [original](https://github.com/mitxela/kicad-round-tracks/issues/32#issue-2900426912) issue. 
[flex-v9 - class params.zip](https://github.com/user-attachments/files/20641227/flex-v9.-.class.params.zip)